srv/viso2 #24: Problem running viso2_ros

aba92, 2015-04-20:
Hello,

When running stereo_odometer, it appears a second robot blinking and with different position and orientation in rviz. That robot also appears in robot camera image, appears a second LaserScan and CostMap. I am not understanding why, and if you could help me I would be very grateful.

miquelmassot, 2015-04-20:
Hi aba92, 

to me, it sounds like there's something wrong with your tf tree. Are you sure there is *only one parent* for each tf frame?

aba92, 2015-04-20:
Yes, I confirmed with rqt_tf_tree. Each frame has only one parent. I leave here a image with what is happening...

[image]

Please note on the cost map, the robot and the second robot on the camera images.

miquelmassot, 2015-04-20:
rqt_tf_tree will only show one parent even if there's more than one. 

Take also into account the time rviz takes to update the images and the model in the window compared to the tf. I have also experienced unsync issues with urdf models and tfs, they update at different rates and sometimes you may see the robot in the camera image frame.

However, on cost maps I don't have any experience. Could you show us your viso2 tf configuration together with everything else? Is there a \base_link frame, an \odom, a \map?

aba92, 2015-04-20:
Yes I have those frames available but I just tried to run the code with _publish_tf:=false and now everything seems to be going fine.

I have another issue (sorry man!), the point cloud output of stereo_odometer is represented in rviz world in a strange direction (z direction) but in the camera images the point cloud can be seen going along the corridor. Can you tell me on what I can be wrong, please.

[image]

miquelmassot, 2015-04-21:
It looks like your camera position is wrong. Check that the *Z* of your camera is pointing towards the corridor, *Y* is pointing downwards and *X* to the right.

I'm closing the issue because this is not a problem related to viso2.

closed 2015-04-21 by miquelmassot