srv/viso2 #26: Using viso2 for monocular odometery

sj312931, 2015-06-23:
Hello,

I am trying to run  monocular odometer for viso2. I am getting the video stream using usb_cam node. I wish to obtain odometry information for my usb cam. I do the following:

Terminal 1:roscore
Terminal 2: rosrun viso2_ros mono_odometer image:=/camera/image_raw
Terminal 3: roslaunch usb_cam.launch
Terminal 4: rosrun rviz rviz

I am not able to see anything in ROS. The usb_cam node is publishing the images on the /camera/image_raw topic. 

Can somebody help me with this? 
Thanks

stwirth, 2015-06-23:
Yes, your camera calibration is the problem. Calibrate your camera first as shown here: http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration

closed 2015-06-23 by stwirth

santhoshrsk, 2017-08-24:
Hi, I am having same issue. I have calibrated my camera properly, but still getting same error.

The tf from '/base_link' to 'camera' does not seem to be available, will assume it as identity!