srv/viso2 #27: running very slow...

antithing, 2015-07-06:
Hi, I have libviso2 running on windows, and it works, BUT is very very slow. Around 3 fps.

I know you just built the ROS port, but do you have any thoughts on why this would be? Is it built to run on the GPU or CPU?

Thanks!

miquelmassot, 2015-07-13:
It's running on CPU. Could you tell us more details about the camera, your hardware and the image resolution you are using?

antithing, 2015-07-13:
Hi, thanks for your reply. I am on windows 8, dual core i7 with 16gb ram. I have a zed camera rig from stereolabs:

https://www.stereolabs.com/

running at 640 x 480 each side.

My graphics card is not the best (nvidia geForce GT 630), but it should be fast enough I would have thought...

When i run the libviso example, using the 2010_03_09_drive_0081 dataset, I see the same speed. 3 - 4 fps.

What frame rate should i be seeing here?

thanks again, appreciate your time.

miquelmassot, 2015-07-13:
I have no experience at all on running viso2 on windows. We mainly work on Ubuntu Linux and ROS. 

With your setup, I doubt that viso2 is the culprit. You may want to take a look to your compilation settings, check if you are compiling with optimization, and if without viso2 you are getting a desirable framerate. Check the disparity image framerate.

antithing, 2015-07-13:
ok thanks! I will try it on another machine and see what i can get. Just for comparison, what frame rate do you get on Linux?

plnegre, 2015-07-20:
With images of 1024x768 we run it at 10Hz.

closed 2015-07-20 by plnegre