srv/viso2 #28: fatal error: viso2_ros/VisoInfo.h: No such file or directory

galaxyeagle, 2015-07-06:
Hi,
I am trying to build libviso in ROS Indigo in my catkin workspace. During build, I keep getting the error : 
/home/raman/catkin_ws/src/dlut_vision/dlut_viso2/dlut_viso2_ros/src/stereo_odometer.cpp:10:32: fatal error: viso2_ros/VisoInfo.h: No such file or directory
 #include <viso2_ros/VisoInfo.h>
                                ^
compilation terminated.
Indeed when I checked the location, there is no VisoInfo.h file . Why is it then included in the stereo_odometer.cpp and mono_odometer.cpp also ?

stwirth, 2015-07-06:
VisoInfo.h gets auto-generated during compilation as it is a ROS message. The problem in your workspace is that you changed the name of the package to dlut_viso2_ros. Rename your package to viso2_ros or replace all reference to viso2_ros with dlut_viso2_ros in the source files.

galaxyeagle, 2015-07-08:
OK, yes changing the dlut_viso2_ros to viso2_ros in the devel folder of the workspace, solved the issue. VisoInfo.h is also auto-generated. Thanks a lot.

closed 2015-07-08 by galaxyeagle