srv/viso2 #30: Changes on the package name

yanming-zou, 2015-07-17:
I am a little confused by the changes on package names. What is the purpose of these changes? Are there any big changes in near future?

I am sorry if this is not the right place for this kind of discussion.

Best regards
Yanming

miquelmassot, 2015-07-17:
Hi Yanming,

which changes are you refering to?

yanming-zou, 2015-07-17:
I mean the package name dlut_viso2. The link in the wiki page of viso2 in ROS also points to another repository. But you are still the maintainer of the package. This situation is confusing to  me. I guess the latest stable version is kept in this repository. And it would be better if there is another wiki page for dlut_viso2 in ROS. What do you think?

stwirth, 2015-07-17:
@ZhuangYanDLUT released viso2_ros viso2 under a synonym. libviso2 is now two times in the rosdistro index, I'll file an issue on https://github.com/ZhuangYanDLUT/dlut_vision.

yanming-zou, 2015-07-20:
Hi Stephan,

I want to try the viso2 package but find the ROS bagfiles link
ftp://opt.uib.es/bagfiles/viso2_ros/ is obsolete. Could you help me to get
some samples? Thanks a lot.

Best regards
Yanming

miquelmassot, 2015-07-20:
We have corrected this issue in #31 . Now bagfiles are at http://srv.uib.es/public/viso2_ros/sample_bagfiles/

closed 2015-07-20 by miquelmassot