srv/viso2 #32: rviz can't subscribe the output message /stereo_odometer/point_cloud

yanming-zou, 2015-07-24:
In the source code, the PointCloud type is defined as pcl::PointCloud<pcl::PointXYZRGB> by
                 typedef pcl::PointCloud<pcl::PointXYZRGB> PointCloud; (line 68, stereo_odometer.cpp)
The compiler thinks it is sensor_msgs::PointCloud2. 
But there are some differences between these 2 types: sensor_msgs::PointCloud2 and pcl::PointCloud<pcl::PointXYZRGB>. 
So rviz can't receive the message when I add the "PointCloud2" display and subscribe "stereo_odometer/point_cloud" in rviz. And the return of point_cloud_pub_.getNumSubscribers() (line 194) in stereo_odometer.cpp is always 0.

I think the above is the reason why I can't show the point cloud in rviz. Am I right? Thanks a lot.

Best regards
Yanming

stwirth, 2015-07-24:
[pcl_ros](http://wiki.ros.org/pcl_ros) does the necessary conversion. There is no message type `pcl::PointCloud<pcl::PointXYZRGB>`. What do `rostopic info /stereo_odometer/point_cloud` and `rostopic hz /stereo_odometer/point_cloud` say?

plnegre, 2015-07-24:
@stwirth is right, ROS makes the convertion between PCL and Points2 (see http://wiki.ros.org/pcl_ros#Publishing_point_clouds). In fact, I'm running viso2_ros just now and I'm able to see /stereo_odometer/point_cloud through the rviz.

stwirth, 2015-07-24:
Closing as not an issue of the library.

closed 2015-07-24 by stwirth

yanming-zou, 2015-07-27:
Thank you all for the help.