srv/viso2 #35: unrecognized command line option '-msse3'

fangthu, 2016-03-22:
hi, I am trying to use this package in my odriod. 

ROS version : indigo 
OS: Ubuntu 14.04

I have downloaded  package in my own workspace. 

When I run 'catkin_make', there are some errors as below: 

Could you help to fix this problem? Thx 

===================================================================
Scanning dependencies of target viso2
[  0%] [  0%] [  0%] [  0%] [  0%] Built target topic_tools_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
Built target sensor_msgs_generate_messages_lisp
Built target nav_msgs_generate_messages_py
Built target roscpp_generate_messages_cpp
Built target sensor_msgs_generate_messages_py
[  7%] [ 14%] [ 21%] Building CXX object viso2/libviso2/CMakeFiles/viso2.dir/libviso2/src/filter.cpp.o
[ 21%] [ 28%] Building CXX object viso2/libviso2/CMakeFiles/viso2.dir/libviso2/src/matcher.cpp.o
[ 28%] Built target _viso2_ros_generate_messages_check_deps_VisoInfo
Building CXX object viso2/libviso2/CMakeFiles/viso2.dir/libviso2/src/matrix.cpp.o
Building CXX object viso2/libviso2/CMakeFiles/viso2.dir/libviso2/src/reconstruction.cpp.o
g++-4.8.real: error: unrecognized command line option '-msse3'
Built target roscpp_generate_messages_py
make[2]: *** [viso2/libviso2/CMakeFiles/viso2.dir/libviso2/src/filter.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
g++-4.8.real: error: unrecognized command line option '-msse3'
make[2]: *** [viso2/libviso2/CMakeFiles/viso2.dir/libviso2/src/matcher.cpp.o] Error 1
[ 28%] g++-4.8.real: error: unrecognized command line option '-msse3'
make[2]: *** [viso2/libviso2/CMakeFiles/viso2.dir/libviso2/src/matrix.cpp.o] Error 1
Built target roscpp_generate_messages_lisp
g++-4.8.real: error: unrecognized command line option '-msse3'
make[2]: *** [viso2/libviso2/CMakeFiles/viso2.dir/libviso2/src/reconstruction.cpp.o] Error 1
make[1]: *** [viso2/libviso2/CMakeFiles/viso2.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 28%] Built target nav_msgs_generate_messages_lisp
[ 28%] Built target actionlib_msgs_generate_messages_cpp
[ 28%] [ 28%] Built target tf_generate_messages_py
[ 28%] [ 28%] Built target actionlib_msgs_generate_messages_lisp
Built target rosgraph_msgs_generate_messages_py
Built target rosgraph_msgs_generate_messages_cpp
[ 28%] Built target nodelet_generate_messages_lisp
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

miquelmassot, 2016-03-22:
Hi @fangthu, this issue looks similar to the one already solved in fovis https://github.com/srv/fovis/issues/17, could you please try the same approach and do a pull request?

fangthu, 2016-03-22:
hi @miquelmassot thank you for quick response. 

I have tried two approach, but it comes with some new issues. 

Approach 1: 

(1). add 'set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_CURRENT_SOURCE_DIR}/cmake/")' to my cmake file 

(2). change ' add_definitions("-msse3")'  to
 'IF(SSSE3_FOUND)
	add_definitions("-msse3")
ENDIF(SSSE3_FOUND)'
(3) create a folder called 'cmake' and create the FindSSE.camke inside it. 

Approach 2:

 (1). add 'set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_CURRENT_SOURCE_DIR}/cmake/")' to my cmake file 

(2). change ' add_definitions("-msse3")'  to
'IF(SSE2_FOUND)
    IF(SSSE3_FOUND)
        add_definitions(-Wall -msse2 -msse3)
    ELSE(SSSE3_FOUND)
        add_definitions(-Wall -msse2)
    ENDIF(SSSE3_FOUND)
ELSE(SSE2_FOUND)
    add_definitions(-Wall)
ENDIF(SSE2_FOUND)'
(3) create a folder called 'cmake' and create the FindSSE.camke inside it. 

The debug information is in the end. It looks like emmintirn,h is wrong. 

My hardware is odriod XU4 with a 2GB RAM .

=============================================================================

[  0%] [  7%] Built target std_msgs_generate_messages_cpp
Building CXX object viso2/libviso2/CMakeFiles/viso2.dir/libviso2/src/filter.cpp.o
[  7%] [  7%] [  7%] [  7%] Built target nav_msgs_generate_messages_py
[ 14%] Built target rosgraph_msgs_generate_messages_lisp
Built target roscpp_generate_messages_cpp
Built target roscpp_generate_messages_lisp
[ 14%] Built target roscpp_generate_messages_py
[ 21%] Building CXX object viso2/libviso2/CMakeFiles/viso2.dir/libviso2/src/matcher.cpp.o
In file included from /home/odroid/Fang/src/viso2/libviso2/libviso2/src/filter.cpp:26:0:
/home/odroid/Fang/src/viso2/libviso2/libviso2/src/filter.h:25:23: fatal error: emmintrin.h: No such file or directory
 #include <emmintrin.h>
                       ^
compilation terminated.
[ 28%] [ 28%] Building CXX object viso2/libviso2/CMakeFiles/viso2.dir/libviso2/src/matrix.cpp.o
Built target topic_tools_generate_messages_cpp
[ 28%] Building CXX object viso2/libviso2/CMakeFiles/viso2.dir/libviso2/src/reconstruction.cpp.o
Built target actionlib_msgs_generate_messages_cpp
[ 28%] [ 28%] Built target geometry_msgs_generate_messages_py
Built target _viso2_ros_generate_messages_check_deps_VisoInfo
[ 35%] Building CXX object viso2/libviso2/CMakeFiles/viso2.dir/libviso2/src/triangle.cpp.o
[ 35%] [ 35%] In file included from /home/odroid/Fang/src/viso2/libviso2/libviso2/src/matcher.cpp:22:0:
/home/odroid/Fang/src/viso2/libviso2/libviso2/src/matcher.h:30:23: fatal error: emmintrin.h: No such file or directory
 #include <emmintrin.h>
                       ^
compilation terminated.
Built target rosgraph_msgs_generate_messages_cpp
[ 35%] Built target std_msgs_generate_messages_py
Built target sensor_msgs_generate_messages_py
make[2]: *** [viso2/libviso2/CMakeFiles/viso2.dir/libviso2/src/filter.cpp.o] Error 1

miquelmassot, 2016-03-23:
It looks like libviso2 needs some compilation flags not supported in ARM platforms. Try removing all sse2/sse3 flags and see if it compiles. 

Please take a look at https://github.com/srv/viso2/issues/25. Have you tried if pale [libviso2](http://www.cvlibs.net/software/libviso/) compiles?

fangthu, 2016-03-23:
hi @miquelmassot   Thank you very much . 
I have tried 'remove all the see2 /see3' as follows: 

1. remove the 'add_definitions("-msse3")'  in the cmake file in libviso2
2. remove the 'add_definitions("-msse3")  'in the cmake file in visor_ros

But is still does not work. 

And when do you mean by 'Have you tried if plain libviso2 compiles' ?  compile this package alone?

Tymolc, 2016-04-11:
Hi. Your odroid XU4 has an ARM processor which indeed doesn't support Intels SSE intrinsics.
You can go into the header files of the libviso2 library and add a "compatibility" header like this one: https://github.com/otim/SSE-to-NEON
You need to replace `#include <emmintrin.h>` and `#include <pmmintrin.h>` (I think in the filter.h and matcher.h) with `#include <SSE-to-NEON.h>` and tell cmake to use compile it with neon capabilites in the CMakeLists.txt (I used these commands: 
```
SET(CMAKE_SYSTEM_PROCESSOR arm)
SET(CMAKE_CXX_FLAGS -mfpu=neon)
```
).
Hope this works.

mesterhazi, 2016-04-20:
Hi! Based on what can you write this compatibility header?
You looked for SSE operations in the code you wanted to compile, and redefined them using the operations in arm_neon.h? Should I look for tutorials and documentations to understand the SSE operations, to be able reconstruct them with neon?
Thanks!

Tymolc, 2016-04-20:
I didn't write the header myself, but that's essentially how it's done.
If you need to convert more SSE intrinsics to ARM there are [a few guides](http://codesuppository.blogspot.de/2015/02/sse2neonh-porting-guide-and-header-file.html) ([Intel released one the other way around](https://software.intel.com/en-us/blogs/2012/12/12/from-arm-neon-to-intel-mmxsse-automatic-porting-solution-tips-and-tricks)), but for viso2 the one I linked is sufficient.

fangthu, 2016-04-20:
thank you very much~  Your suggestions are really helpful. @miquelmassot @Tymolc 
I have figure it out by following you two gays' ideas. I will close this issue soon.  :)

miquelmassot, 2016-04-21:
Thanks @fangthu, @Tymolc! If finally viso2 works in ARM, make a pull request and I'll merge it in another branch so that anyone interested can use it.

syamprasadkr, 2016-06-13:
@ Tymolc
I am trying to install viso2_ros package on Jetson TX1, running ROS Indigo on Ubuntu 14.04.4. Here is what I did (I am pretty new to ROS and Ubuntu):
I downloaded SSE-to-NEON.hpp file from the link given and pasted it in the folder: 

> /home/ubuntu/catkin_ws/src/viso2/libviso2/libviso2/src/

Then I edited filter.h and matcher.h to include this file. Then I edited the CMakeLists.txt file located at:

> /home/ubuntu/catkin_ws/src/viso2/libviso2/libviso2
I added:
 `SET(CMAKE_SYSTEM_PROCESSOR arm)
SET(CMAKE_CXX_FLAGS -mfpu=neon)
#SET(CMAKE_CXX_FLAGS "-msse3")`

I also edited the CMakeLists.txt located at:

> /home/ubuntu/catkin_ws/src/viso2/libviso2
I added and edited lines as:
`SET(CMAKE_SYSTEM_PROCESSOR arm)
SET(CMAKE_CXX_FLAGS -mfpu=neon)
#set(CMAKE_CXX_FLAGS "-msse3")
add_definitions(-mfpu=neon)`

Now when I run catkin_make, I get  error saying mm_free and mm_malloc are not declared in scope. How can I fix this? Thank You.

Tymolc, 2016-06-13:
Right, I think I had the same problem. What I did was, I commented the check if MM_ALLOC_H is defined (line 14 and corresponding endif in line 46 in the sse-to-neon.h)
I don't know if that's a bad idea and if it can break anything, but it worked for me so far.

syamprasadkr, 2016-06-13:
@Tymolc 
Thank You so much. It worked and the package compiled successfully. I hope I haven't broken anything.

miquelmassot, 2016-06-15:
If you want to provide a pull request I would merge it to a new ARM branch for future users.

syamprasadkr, 2016-06-15:
@miquelmassot I would love to help, but I am quite new to ROS. So, I would like to know how you want it to be done. I know what files I have edited. Do you want me to provide just those files or do you want me copy the entire viso2 folder from my ROS Indigo catkin workspace? I can also include a readme file inside the folder.

miquelmassot, 2016-06-16:
@syamprasadkr I meant through a pull [request](https://help.github.com/articles/using-pull-requests/)

I forked our own project on my github and did the changes you mentioned. Could you please confirm? The fork is in https://github.com/miquelmassot/viso2 branch **arm**.

Quinche78, 2017-01-26:
Hi, I am trying to install viso2_arm package on Firefly 3288 plus board, running ROS Indigo, but when I run catkin_make, I get error. I am deletes all lines before mentionated (-msse3) but can not works. Which is the posible problem?. Thanks you.

gbr1, 2017-11-12:
@miquelmassot I tried to install your "arm" branch but when compiling I got the -msse3 option error :(
I'm using ROS kinetic on Ubuntu 16.04 on Odroid XU4, could you help me?

miquelmassot, 2017-11-13:
Hi @gbr1, have you tried the solution proposed in [this comment](https://github.com/srv/viso2/issues/35#issuecomment-208262779)?

gbr1, 2017-11-13:
@miquelmassot yes!
I cloned this [one](https://github.com/srv/viso2/pull/43/files) that already contains modifications.
I also make modifications as @Tymolc wrote (comment line 14 and 46).
It's really strange :(

miquelmassot, 2017-11-14:
Have you tried also this one? https://github.com/miquelmassot/viso2/tree/arm

gbr1, 2017-11-14:
@miquelmassot Yes!
This is the screenshot :(
[image]

Maybe could be this issue only for odroid? ( @syamprasadkr used a TX1)

Tymolc, 2017-11-14:
Can you make sure that Neon capabilities are enabled (i.e. the #if defined(__ARM_NEON__) is true)?
Someone in [this](https://stackoverflow.com/questions/33045812/compiling-neon-code-on-an-odroid-xu4) stackoverflow thread used some additional CMake switches (including -DENABLE_NEON) on the same odroid that you use. Maybe that's worth a try.

gbr1, 2017-11-14:
@miquelmassot @Tymolc 

`cmake_minimum_required(VERSION 2.8.3)`
`project(libviso2)`

`find_package(catkin REQUIRED )`

`#set(CMAKE_CXX_FLAGS "-msse3")`
`#add_definitions(-msse3)`

`#option(ARM_CROSS_COMPILATION "ARM Cross Compilation" OFF)`

`set(NEON_FLAGS "-DENABLE_NEON -mfloat-abi=hard -mfpu=neon-vfpv4 -mcpu=cortex-a1$`
`set(CMAKE_CXX_FLAGS "-std=c++0x ${CMAKE_CXX_FLAGS} -Wno-format-security ${NEON_$ ...`
`set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} ${NEON_FLAGS}")`

`#if(ARM_CROSS_COMPILATION)`
`#  SET(CMAKE_SYSTEM_PROCESSOR arm)`
`#  SET(CMAKE_CXX_FLAGS -mfpu=neon)`
`#endif(ARM_CROSS_COMPILATION)`

edited in this way, commented in **sse_to_neon** line 14 and 46, now achieve 98%, I think this is the way!
[image]

I edited also CMakeList in viso2_ros but it seems not work... :(

fabolhak, 2018-04-16:
Hello,
thank you for all the tips. I also got it working on our Jetson TX2. My changes can be found here:
https://github.com/fabolhak/viso2/commit/6816970aee109dfd284d4fd16a62586454f2b17b

Let me know if I should create a pull request.

Shantanu-Karnwal, 2018-04-27:
@fabolhak Hi, we did try cloning your branch to our odroid and it is showing us the same error as before. Are we missing something? Can you please check? Its urgent.

fabolhak, 2018-04-28:
Hey, what kind of error are you referring to? Still the `unrecognized command line option '-msse3'` error? Can you recheck if you are on branch `arm` and have arm cross-compilation option enabled (`catkin_make --cmake-args -DARM_CROSS_COMPILATION=ON`).

This is the only possible errors I can think of. Unfortunately, I don't have an odroid here (only Jetson).

Shantanu-Karnwal, 2018-04-28:
Yeah, checked. Doesn't work. Do you think there might be something else that makes it different for your code to run on Jetson compared to the odroid. If there is I might be able to check and make it run for the odroid.

papr, 2018-06-18:
I was able to find all the places that set the `-msse3` flag and change them appropriately. Please see this PR for details and discussion: https://github.com/miquelmassot/viso2/pull/1

closed 2018-06-25 by miquelmassot