srv/viso2 #36: Visual odometry lost and Segmantation fault

siddyyagno3, 2016-04-01:
This is the error which occurs during runtime of viso2. Camera is calibrated using camera_calibration cameracalibrator.py .
Camera: Logitech c270
Camera Driver: uvc_camera

sidd@sidd-K55VM:~/catkin_ws$ rosrun viso2_ros stereo_odometer _approximate_sync:=True  stereo:=stereo image:=image_rect
[ INFO] [1459528291.959742716]: Subscribing to:
	* /stereo/left/image_rect
	* /stereo/right/image_rect
	* /stereo/left/camera_info
	* /stereo/right/camera_info
[ INFO] [1459528292.086546853]: Basic Odometer Settings:
  odom_frame_id      = /odom
  base_link_frame_id = /base_link
  publish_tf         = true
  invert_tf          = false
[ INFO] [1459528292.534461262]: Initialized libviso2 stereo odometry with the following parameters:
Calibration parameters:
  f  = 953.869
  cu = 679.641
  cv = 239.209
Matcher parameters:
  nms_n                  = 3
  nms_tau                = 50
  match_binsize          = 50
  match_radius           = 200
  match_disp_tolerance   = 2
  outlier_disp_tolerance = 5
  outlier_flow_tolerance = 5
  multi_stage            = 1
  half_resolution        = 1
  refinement             = 1
Bucketing parameters:
  max_features  = 2
  bucket_width  = 50
  bucket_height = 50
Stereo odometry parameters:
  base             = 0.386787
  ransac_iters     = 200
  inlier_threshold = 2
  reweighting      = 1
  queue_size = 5
  approximate_sync = 1
  ref_frame_change_method = 0
  ref_frame_motion_threshold = 5
  ref_frame_inlier_threshold = 150
[ WARN] [1459528292.586841816]: The tf from '/base_link' to 'camera' does not seem to be available, will assume it as identity!
[ WARN] [1459528292.722457090]: Visual Odometer got lost!
[ WARN] [1459528302.841250915]: Visual Odometer got lost!
[ WARN] [1459528312.914954366]: Visual Odometer got lost!
[ WARN] [1459528323.099015192]: Visual Odometer got lost!
[ WARN] [1459528333.212741587]: Visual Odometer got lost!
[ WARN] [1459528343.409453980]: Visual Odometer got lost!
[ WARN] [1459528353.778834532]: Visual Odometer got lost!
[ WARN] [1459528363.782220476]: Visual Odometer got lost!
[ WARN] [1459528373.993243490]: Visual Odometer got lost!
[ WARN] [1459528384.164382081]: Visual Odometer got lost!
Segmentation fault (core dumped)

We are publishing image_rect and the Odometer is subscribing to it and camera_info. We can't seem to figure out why we are getting a segfault.

bane123, 2017-07-05:
have you find the solution to your problem @siddyyagno3 .I have same issue , can u help me ?
Thanks in advance .

miquelmassot, 2017-07-07:
Hi @siddyyagno3  and @bane123, thanks for reporting. Can you please give us more information, such as a GDB backtrace and system info?

bane123, 2017-07-19:
Thanks for considering @miquelmassot 
I got solution to my problem , I was not doing transformation and video i was playing has high frame rate . I slow down the video and did static transformation it resolved my problem.

closed 2018-06-25 by miquelmassot