srv/viso2 #37: Pose change without camera movement

niteeshsood, 2016-04-11:
I'm running the following configuration.

Two logitech c270 cameras separated by ~14cm.
After camera calibration we get an epipolar error of 0.70 pixels approximately. 
I'm getting a reasonable point cloud and am including photos of the configuration.

[image]

I am checking the pose using a node which is making a log file. Here's the code for it.

```
#!/usr/bin/env python
import os
import rospy
import rospkg
import numpy as np
#from python_qt_binding import loadUi
#from python_qt_binding.QtGui import QWidget
#from python_qt_binding.QtCore import QTimer, Slot
#from svo_msgs.msg import Info
#from svo_msgs.msg import Feature
#from std_msgs.msg import String
from geometry_msgs.msg import PoseWithCovarianceStamped, Pose, PoseStamped
import geometry_msgs

from datetime import datetime
logs_path = '/home/niteesh/viso2logs'
  

ns = 'stereo_odometer'

def i_cb(data,fw):
  rospy.loginfo('Viso2 is tracking')
  #posestampd = data.pose
  pose_ = data.pose
  posep = pose_.position
  x = posep.x
  y = posep.y
  z = posep.z
  rospy.loginfo('x:%f, y:%f, z:%f' % (x, y, z) )
  fw.write('x:%f, y:%f, z:%f\n' % (x, y, z))
  

def listener():
    rospy.init_node('pose_listener', anonymous=True)
    rospy.loginfo('Trying to subscribe')
    dt_str = str(datetime.now())
    logfile = os.path.join(logs_path, dt_str[0:10]+'-'+dt_str[11:13]+'-'+dt_str[14:16]+'-'+dt_str[17:19]+'.txt')
    fw = open(logfile, 'w')
    try:
      rospy.Subscriber(ns+'/pose', PoseStamped, i_cb, fw)
    except Exception,e:
      rospy.loginfo(e)

    rospy.loginfo('Not getting any information from topic')
    # spin() simply keeps python from exiting until this node is stopped
    rospy.spin()
    

if __name__ == '__main__':
    listener()
```

Without any movement of the camera, the logger shows that the x coordinate increases with time. The odometer does not get lost either.

miquelmassot, 2016-04-12:
Hi @niteeshsood! 

It is normal that the odometry drifts a little. However, I think there's something to it: In my opinion, the *min_disparity* param has a high value. Do the cloud points make sense? I mean, do you get the same distance from the camera to that flag than from the camera origin to the cloud in Rviz? 

If not, try lowering the *min_disparity* to 1-5 pixels, and the disparity range to the lowest value you can and see what you get. It seems that the flag has a repeating texture and the matcher may fail. Do you have a non-repeated textured object? A journal cover, for instance.

niteeshsood, 2016-04-12:
Hi @miquelmassot thanks for your reply.

I managed to stop the drifting by restarting and recalibrating the camera. But what you say makes sense. I will try with a non-repeating texture and see what results I get and follow up on this.

miquelmassot, 2016-04-12:
ok @niteeshsood! I'll close the issue then. Feel free to open it again if needed.

closed 2016-04-12 by miquelmassot