srv/viso2 #39: mono vo gets poor velocity estimate

fangthu, 2016-04-22:
hi .I have got the mono vo worked on my odroid. 
But the result seems to be poor especially when the vehicle is turning.  

I have done the camera calibration as the instruction. 
I also set the camera pitch, height, the tf from base_link to camera. 
I don't set other parameters. 
I put a lot of pictures on ground to add more features. 

The result of vo is only 1.8 hz. The rate of images which feed into the mono vo is 30Hz. 

I wonder what else i can do to improve the performance of mono vo? like how to tuning the parameters? 

And if the poor performance is due to the low frequency?

plnegre, 2016-04-22:
Hi @fangthu! What is the size of the images?

fangthu, 2016-04-22:
@plnegre  the image is 640*480. fps = 10.

miquelmassot, 2016-06-15:
It looks like you PC is not capable of processing your 10 (or 30?) fps, and it is dropping frames. Therefore the poor performance.

If you have any other question, feel free to reopen this issue.

closed 2016-06-15 by miquelmassot