srv/viso2 #41: How to install with apt-get and invoking "make" -j8 -l8 " failed

Tondy, 2016-05-20:
Hello

How can some one install this package using apt-get have tried : sudo apt-get install ros-indigo-viso2 no success. Also when I download git into src and run catkin make.

invoking "make" -j8 -l8 " failed appears.

Hope you can help.

miquelmassot, 2016-05-23:
Hi @Tondy! Thanks for reporting.
We are not providing ROS binaries, so it's normal that you cannot find ros-indigo-viso2. However, you should be able to compile it from source. Could you please tell us which OS, ROS distribution and git branch are you using? Could you also paste the error so that I can further help you? You may paste the full output in pastebin (http://pastebin.com/) and send us the link.

Tondy, 2016-05-28:
Hello Mique

I have not written any code. I git cloned viso2 into my src file in ROS. When I run catkin_make in the main catkin folder the error  "make" -j8 -l8 " failed when trying to install packages happen. I have ros indigo and a intel nuc as the computer.

miquelmassot, 2016-05-28:
Can you show me the full output error so that I can further help you? You may copy&paste the full output in pastebin (http://pastebin.com/) and send here the link.

closed 2018-06-25 by miquelmassot