srv/viso2 #46: Works great with recorded images, but not with live...

antithing, 2017-02-17:
Hi, I have a realsense r200 camera, and am using it with Stereo Odometry.

The images are converted to cv::mat like this:

`Mat lFrame(480, 640, CV_16U, (uchar *)dev->get_frame_data(rs::stream::infrared));`

which looks fine, and is from the docs.

When i save images out using:

`cv::imwrite(name, img);`

and run them through the libviso by loading the images, like this:

```
cv::Mat left_img = cv::imread(left_img_file_name.c_str(), 0); // load as black and white image
uint8_t* left_img_data = left_img.data;
```

Everything works great. BUT, when i use a live stream of images, like this:

```
Mat lFrame(480, 640, CV_16U, (uchar *)dev->get_frame_data(rs::stream::infrared));
uint8_t* left_img_data = lFrame.data
```

It does not work at all.

I suspect this is due to a mismatch in Mat types, but I am lost.

I have tried:
```

cv::cvtColor(lFrame, leftImg, CV_GRAY2RGB);

cv::cvtColor(lFrame, leftImg, CV_GRAY2BGR);
```
But I see no difference. Does anyone have any thoughts on what might be missing here?

miquelmassot, 2017-02-17:
Hi @antithing! 

have you checked the output odometry frequency vs input frequency of the stream? I suspect it may be a CPU computing time issue.

antithing, 2017-02-17:
Hi! Thanks for getting back to me.  Do you mean I just need to add a delay before the next frame is loaded? Where can i check the output frequency?

miquelmassot, 2017-02-18:
I assume you are using ROS, use _rostopic hz_ command.

antithing, 2017-02-19:
Thanks again, it turns out I just needed to convert my images to 8bit before i stored the data. :)

closed 2017-02-19 by antithing

nmarticorena, 2020-01-10:
> Hi, I have a realsense r200 camera, and am using it with Stereo Odometry.
> 
> The images are converted to cv::mat like this:
> 
> `Mat lFrame(480, 640, CV_16U, (uchar *)dev->get_frame_data(rs::stream::infrared));`
> 
> which looks fine, and is from the docs.
> 
> When i save images out using:
> 
> `cv::imwrite(name, img);`
> 
> and run them through the libviso by loading the images, like this:
> 
> ```
> cv::Mat left_img = cv::imread(left_img_file_name.c_str(), 0); // load as black and white image
> uint8_t* left_img_data = left_img.data;
> ```
> 
> Everything works great. BUT, when i use a live stream of images, like this:
> 
> ```
> Mat lFrame(480, 640, CV_16U, (uchar *)dev->get_frame_data(rs::stream::infrared));
> uint8_t* left_img_data = lFrame.data
> ```
> 
> It never succeeds at:
> 
> ` if (viso.process(left_img_data,right_img_data,dims)) {`
> 
> and jumps straight to:
> 
> ```
> } else {
>         cout << " ... failed!" << endl;
>       }
> ```
> 
> I suspect this is due to a mismatch in Mat types, but I am lost.
> 
> I have tried:
> 
> ```
> 
> cv::cvtColor(lFrame, leftImg, CV_GRAY2RGB);
> 
> cv::cvtColor(lFrame, leftImg, CV_GRAY2BGR);
> ```
> 
> But I see no difference. Does anyone have any thoughts on what might be missing here?

Hi,excuse me for bother you, how do u manage to run the stereo odometry version on the realsense?