srv/viso2 #47: Memory performance

dawnos, 2017-03-14:
Hi guys!
I've been using libviso2 for a while. Recently I found that the ROS wrapper is memory consuming. The memory increased continuously when I ran viso2_ros. But to my knowledge, visual odometer should not behave like this. Is it storing something else? Do any body meet this problem?

seanbai2008, 2017-04-13:
Hi Dawnos,

I'm having exactly same problem. Now I'm using ZED camera to feed video to viso2 and when I run libviso2, my memory keep increasing. I did find that my viso2 was using Opencv3 before and I switched it back to opencv2. however the problem is still there.

dawnos, 2017-04-14:
Have you recompiled viso2? Which version of opencv2 do you use(by apt-get or compile by yourself)?
As I know, zed SDK requires opencv3. Have you uninstall opencv3 totally?

seanbai2008, 2017-04-14:
Thanks for reply. Yes I did some changes in my CMakelist in viso2 and
recompiled it using catkin. I check in the build directory that it is
linked to opencv2.4.8. Do I really need to uninstall opencv3? I have
virtual environment for opencv3 and local version of opencv is 2.4.8. They
are mutually independent.

dawnos, 2017-04-14:
You can use `ldd executable` to see which version of opencv does your viso2 link to actually. If it uses opencv2, it may be some problem else.