srv/viso2 #50: package interferes with pointgrewy cameras

benjaminabruzzo, 2017-07-20:
I've spent that last week trying to figure out why my pointgrey cameras stopped publishing their camera_info topics.  It happens immediately after installing this package.

I'll be happy to supply config info, but the short story is this:
ubuntu 14.04.5
ros indigo
pointgrey-camera-driver (2x chameleon cm3-u3-13s2c)

The whole pipeline that I follow to install (sanitized for paths and licenses)
[pipeline.txt](https://github.com/srv/viso2/files/1160947/pipeline.txt)

miquelmassot, 2017-07-20:
Where did you get viso2 from?

    mkdir ~/ros/src/viso2-indigo && cd ~/ros/src/viso2-indigo
    git init
    git remote add bb [path-to-package]
    git pull bb master && catkin_remake

benjaminabruzzo, 2017-07-20:
I started here: http://wiki.ros.org/viso2_ros , then imported a copy of the [indigo tree](https://github.com/srv/viso2/tree/indigo) to bitbucket, which I use to keep code versions frozen.

miquelmassot, 2017-07-20:
You wrote it happened after installing. Even if the nodes are not used. Can you show us a minimal example launchfile?

benjaminabruzzo, 2017-07-21:
I'm working on a minimal launch file as well as a minimal clean install.  I'll put it here when I finish.

benjaminabruzzo, 2017-07-28:
I believe the issue was with the pointgrey camera driver.  There is a known issue with the camera_info not publishing on ros-indigo installations when using usb cameras.
[issue 137](https://github.com/ros-drivers/pointgrey_camera_driver/issues/137)
[issue 134](https://github.com/ros-drivers/pointgrey_camera_driver/issues/134)

closed 2017-07-28 by benjaminabruzzo