srv/viso2 #51: Cannot build viso2

kobylyanskiy, 2017-08-01:
I've tried to build viso2, installed all dependencies, but still are getting error 'Invoking "make -j4 -l4" failed' while building stereo_odometer. Before it there was an error  ‘shared_ptr’ in namespace ‘std’ does not name a template type. Could you help me, please? Is it a problem with image_geometry or something else?

miquelmassot, 2017-08-25:
Hi @kobylyanskiy,

I think you may have downloaded a recent version of viso2 which is using c++11 shared_ptr. Can you check which Ubuntu, ROS and viso2 branch are you using?

kobylyanskiy, 2017-08-28:
Hi @miquelmassot,

Thank you for your response. I've already solved this issue. I downgraded to the older commits and everything was built. I was using Ubuntu 16.04, ROS keenetic and branch master.

raoshashank, 2017-10-12:
Which version and commit sha worked for you @kobylyanskiy ? I am facing the same problem currently on ROS indigo ubuntu 14.04. I am getting the same error as yours.

closed 2018-06-25 by miquelmassot