srv/viso2 #52: viso2_ros cannot obtain the reconstruction

walkeryu123, 2018-01-04:
Hi, 
I am using viso2_ros with ROS indigo,  and the demo.launch seems work. 
However when I use rviz with PointCloud2(stereo_odometer/point_cloud), I get the Transform error: For frame [stereo_forward_optical]: Frame [stereo_forward_optical] does not exist. 

[image]

Can you give me some advice to solve this problem? Thanks in advance.

plnegre, 2018-01-04:
Can you run:
```
rosrun tf view_frames
```
and post the tf tree here?

walkeryu123, 2018-01-04:
@plnegre  Thanks for your help. 

[image]

Sorry, I do not know how can I change stereo_forward_optical to  stereo_forward

plnegre, 2018-01-04:
What is the frame_id of your images? stereo_forward or stereo_forward_optical? Check it in the header of your image topic.

walkeryu123, 2018-01-04:
@plnegre  I use test.bag downloaded from http://srv.uib.es/public/viso2_ros/sample_bagfiles and demo.launch in viso2_ros.  I do not modify these files. So I think the  frame_id is stereo_forward and I do not know why rviz need stereo_forward_optical.
[image]

plnegre, 2018-01-04:
Ok, so in the demo.launch change this:

```
<param name="base_link_frame_id" value="$(arg camera)"/>
```

for this:

```
<param name="base_link_frame_id" value="stereo_forward_optical"/>
```

walkeryu123, 2018-01-04:
@plnegre  It works. Thank you for your help!

closed 2018-01-04 by walkeryu123