srv/viso2 #66: Release for ROS Melodic

RonaldEnsing, 2019-02-22:
Is there any plan to release viso2 for ROS Melodic?

FPSychotic, 2019-03-06:
please do it,,,, that should allow make work realsense cameras as stereo cameras, now, we are forced to use rgb which is very power demand

RonaldEnsing, 2019-03-11:
Although there is a wiki page for viso2 on ROS Indigo [1], I can only find a doc entry in rosdistro [2]. It looks like there is no released version on any distro. Any plans for doing an release? i think it would be really useful.

[1] http://wiki.ros.org/viso2_ros
[2] https://github.com/ros/rosdistro/blob/master/indigo/distribution.yaml#L17215

knorth55, 2021-05-10:
+1 for this issue.
We are looking forward to installing this package from `apt`