srv/viso2 #69: how to apply on carla-ros-bridge

slinshady89, 2019-06-16:
Hello I'm struggling to use libviso with my available topics. I want to use stereo_odometry on the right and left camera which coordinate systems are already parallel and only a translation in y in vehicle coordinates is applied. However after 'catkin_make' I neither find rosun stereo_odometer nor any of the other proposed packages. However, when I do 'catkin_make' they seem to be build. 

[image]

[image]

ysong-aceinna, 2019-06-18:
Hi, here is a demo.launch of how to use stereo_odometer, hope it is useful.
(https://github.com/srv/viso2/blob/25db160641faa5a1cb09147716d15dbba86c6bd5/viso2_ros/launch/demo.launch)

slinshady89, 2019-06-18:
I was able to use it and it starts now but I have an instant segmentation fault
[image]

ysong-aceinna, 2019-06-18:
I used a bag file which was downloaded from [here](http://srv.uib.es/public/viso2_ros/sample_bagfiles/), and didn't encounter segmentation fault, maybe you can have a try.

slinshady89, 2019-06-18:
I found out that if I add some roll to one of the cameras it's not compensated in image_rect by `stereo_image_proc`. Probably the `base_link` isn't established correctly.
[image]
that's the transformation done by `rosrun tf tf_echo ego_vehicle ego_vehicle/camera/rgb/left`
[image]
But I think I get my segmentation fault because the tf couldn't be done and is to contrary to the identity matrix. left or right frame is no difference
[image]

klintan, 2019-09-10:
Pretty sure it is the same issue as this one https://github.com/srv/viso2/issues/63#issuecomment-530021716

as @ysong-aceinna mentions, using the height and width of the bag-files images will most likely fix the issue.