srv/viso2 #70: questions of running launch file

ysong-aceinna, 2019-06-22:
Hi, I think there is a typo in README.md about the launch files name.
[image]

Besides, I have two questions as below and look forward your great help.
Question 1: when I used performance_test.bag and ran "roslaunch viso2_ros omni_version.launch",  process died and feedback "ERROR in SVD: No convergence in 30 iterations".
[image]

Question 2: when I used performance_test.bag and ran "roslaunch viso2_ros perspective_version.launch",  position and attitude from topic "/mono_odometer/odometry" have big drift, as the green trajectory of below picture.
[image]

Thanks for you help in advance.

aaguiar96, 2019-06-23:
Hello!
About the README you're write. We will fix that issue.

*Question 1:* For you to use the perspective camera in the omnidirectional version you'll have to calibrate it using the omnidirectional camera calibration toolbox mentioned in the README file. The SVD error is due to the use of the calibration file reference to the fisheye camera of the bag file instead of a proper calibration for the perspective one.
However, there is no advantage on using the perspective camera in the omnidirectional version. This version intends to solve the high radial distortion present in dioptric cameras...

*Question 2:* You can try to tune the parameters in order to get a better performance. However, the conditions where the bag file was recorded are challenging (slow motion and high turbulence) and as this is a monocular VO method, the scale calculation is an inherent issue.

ysong-aceinna, 2019-06-24:
Hi @aaguiar96 , thanks for your responding!

For question 1:  yes you are right, it works normally after I set the correct calibration file path in launch file, and I got trajectory from rostopic "/mono_odometer/odometry" as below. Similar big drift as I mentioned in question2.
[image]

Except monocular VO method, I also tried the **stereo** VO method and met similar big position and attitude drift which mentioned in [issue 68](https://github.com/srv/viso2/issues/68). Except the test.bag developer provided, I also made a few bagfiles which generated from [Kitti datasets](http://www.cvlibs.net/datasets/kitti/raw_data.php)  to test the stereo VO method, when I found the big drift issue, I referred to a close [issue 45](https://github.com/srv/viso2/issues/45#issuecomment-240418551) and set 
`~ref_frame_change_method = 2   
 ~refinement = 2`
 , but I didn't get an obvious performance improvement.

I see the scale calculation is an inherent issue of algorithm but not the wrapper, it is also greatly appreciated if you have any parameters suggestion to improve the stereo VO performance according  to your experience.

Thank you very much for your input.